Northeastern-Electric-Racing/shepherd_bms #18: Streamline debugging for state machine

nwdepatie, 2022-09-16:
This ticket is broad, but relates to making the state machine easier to debug/fix if something goes awry and we need to fix it quickly. This might include creating fault enumerators, ways of outputting what state we are in if we fault, etc

dyldonahue, 2023-02-10:
Here's the scoop: 

if something goes wrong in our code, the code will stop executing somewhere. For right now (though we can make it more robust in the future), simply placing serial prints in tactical locations throughout the code can help us figure out where the code stops executing. 

For this ticket, place a bunch of these prints throughout segment related stuff (segment.cpp) and make them descriptive so that we know where each one corresponds to.

Also, we don't want these executing all the time, only when we debug. To fix this, add ifdef guards. (to see an example of how this works, check main.cpp, and platformio.ini to see how those macros are initialized)

dyldonahue, 2023-02-10:
Here's the scoop: 

if something goes wrong in our code, the code will stop executing somewhere. For right now (though we can make it more robust in the future), simply placing serial prints in tactical locations throughout the code can help us figure out where the code stops executing. 

For this ticket, place a bunch of these prints throughout statemachine related stuff (Statemachine.cpp) and make them descriptive so that we know where each one corresponds to.

Also, we don't want these executing all the time, only when we debug. To fix this, add ifdef guards. (to see an example of how this works, check main.cpp, and platformio.ini to see how those macros are initialized)

closed 2023-05-17 by dyldonahue