Northeastern-Electric-Racing/shepherd_bms #19: Implement CRC error checking

nwdepatie, 2022-09-16:
This can either be actually adjusting the message based on the CRC, or just disregarding the data. Make sure the LTC6820 doesn't already do this

airman416, 2022-10-13:
From what I can see the PEC is [already implemented in the LTC library](https://github.com/Northeastern-Electric-Racing/shepherd_bms/blob/fbb2e22cbcdc27992c9ec4a0042767b8b9373010/shepherd_bms/lib/LTC/LTC68041.cpp#L945). It seems to me that if there is a PEC error the device will just [retry read](https://github.com/Northeastern-Electric-Racing/shepherd_bms/blob/fbb2e22cbcdc27992c9ec4a0042767b8b9373010/shepherd_bms/lib/LTC/LTC68041.cpp#L262), so there is nothing to do on our end... correct me if I'm wrong

nwdepatie, 2022-10-18:
After perusing the code, I came across the fact that everything after line ```1046``` in ```LTC68041.cpp``` is actually made by @nvelcea and @zoomiti, so there is definitely some work to optimize/implement PEC there.

Example implementation of the PEC calc is here:
https://github.com/Northeastern-Electric-Racing/shepherd_bms/blob/fbb2e22cbcdc27992c9ec4a0042767b8b9373010/segment_testing/Test_Fixture_3/src/LinduinoFiles/LTC68041.cpp#L1158

Here is an example of somewhere the PEC calc needs to go:
https://github.com/Northeastern-Electric-Racing/shepherd_bms/blob/fbb2e22cbcdc27992c9ec4a0042767b8b9373010/shepherd_bms/lib/LTC/LTC68041.cpp#L1121

nwdepatie, 2022-10-18:
Now there are two courses of action that we can take:
1. Just return -1 if the PEC's don't match and ignore the reading in the SegmentInterface
2. If the PEC's don't match, we might want to just have the function automatically retry a couple times up until a timeout, then return -1 if it *really* can't get data

I think either would work as long as we implement them correctly

closed 2023-05-17 by dyldonahue