Northeastern-Electric-Racing/shepherd_bms #75: Implement better CCL and DCL calculations

nwdepatie, 2022-12-15:
**What are you proposing?**
We need to factor in voltage and any other factors into the DCL and CCL calculations

**What files/components does this concern?**
probably ```calcs.h``` and ```calcs.cpp```

**What would this improve over how it is now?**
Make the CCL and DCL more accurate

**Estimated time commitment**
Maybe like a couple hours, we need to research how and then implement

mccauleyma, 2023-01-13:
DCL is done! No longer high priority, but CCL should maybe also be done? Should at least assess it and see if Orion does it.

dyldonahue, 2023-02-15:
@mccauleyma are u content with everything in this ticket or should i keep it open to revisit

mccauleyma, 2023-02-15:
CCL should be done probably

closed 2023-06-10 by dyldonahue