Northeastern-Electric-Racing/shepherd_bms #86: Implement thermistor averaging

mccauleyma, 2023-01-13:
Thermistors have a lot of weirdness and/or noise. We can implement a moving average to smooth this. Hopefully would fix the issue of random 13C & 33C readings

mccauleyma, 2023-01-13:
#102 is including the addition of an incremental smoothing. Attempted averaging but it did not work. Seems that we have a loose connection in the battbox that we are trying to fight with formware. For now we can call it good and then revisit after rebuild

mccauleyma, 2023-01-22:
@nwdepatie Why did this get assigned out and there is a PR? As mentioned about I think it would be best to not mess with this until we rebuild the battbox and fix hardware issues so we can isolate this to software

dyldonahue, 2023-01-31:
on hold until above comment is addressed. PR that was made and closed can be reopened and visited at that time

closed 2023-03-14 by dyldonahue