Northeastern-Electric-Racing/shepherd_bms #106: Need to add fractional current data for low range sensor

dyldonahue, 2023-01-15:
The current sensor has a high precision range in which we should store current with decimal accuracy. This would be used for better open cell voltage calculation and charging

nwdepatie, 2023-01-21:
@mccauleyma Is this done already? It looks like the low channel already is a float

mccauleyma, 2023-01-22:
Not that im aware of. Possible that I swapped something to be a float and then forgot to properly implement it later. Current should be an integer that is current * 10 to represent current with single decimal precision

dyldonahue, 2023-01-31:
@mccauleyma to somewhat reiterate @nwdepatie, right now the low channel current is calculated as a float, but then is assigned to an integer when its read in. Does this mean all u want us to do is to convert that float to an int and multiply this by 10?

```
static const float current_lowChannelMax = 75.0; //Amps
    static const float current_lowChannelMin = -75.0; //Amps
    static const float current_supplyVoltage = 5.038;
    static const float current_ADCResolution = 5.0 / MAX_ADC_RESOLUTION;

    static const float current_lowChannelOffset = 2.530; // Calibrated with current = 0A
    static const float current_highChannelOffset = 2.57; // Calibrated with current = 0A

    static const float highChannelGain = 1 / 0.0040;
    static const float lowChannelGain = 1 / 0.0267;

    uint16_t highCurrent = (5 / current_supplyVoltage) * (analogRead(CURRENT_SENSOR_PIN_H) * current_ADCResolution - current_highChannelOffset) * highChannelGain; // Channel has a large range with low resolution
    float lowCurrent = (5 / current_supplyVoltage) * (analogRead(CURRENT_SENSOR_PIN_L) * current_ADCResolution - current_lowChannelOffset) * lowChannelGain; // Channel has a small range with high resolution
```

mccauleyma, 2023-01-31:
Multiply by 10 first, then convert, yes.

I kinda forgot this part was already floats, so yeah ig it is just multiply by 10, concert to int, and then just update the rest of the code to use ints that are 10x

closed 2023-02-09 by dyldonahue