Northeastern-Electric-Racing/shepherd_bms #110: Look into redoing Open Circuit Voltage

nwdepatie, 2023-01-15:
**What is going wrong?**
Right now, the Open Circuit Voltage is annoying to calculate because the motor will draw a bit of power even if we have come to a complete stop, we need to wait for the car to completely settle before setting the open circuit voltage to the current voltage reading.

We might need to figure out a better way to conditionally set open circuit voltage

This came about because when the motor controller boots, it results in a discharge current limit fault

dyldonahue, 2023-02-09:
@nwdepatie @mccauleyma is this worth investigating if the workaround to avoid the fault continues to work?

mccauleyma, 2023-02-11:
If that fix I made works then yeah, this may not need further improvement. The current threshold should be further tuned though. We need to raise the threshold such that the motor controller draw doesn't stop open cell calcs (otherwise we woould only recalculate when the tractive system is fully off).

dyldonahue, 2023-03-14:
closing because redundant based on other tickets. If the fix matt implemented doesnt work, will go to drawing board to decide if this approach is best

closed 2023-03-14 by dyldonahue