Northeastern-Electric-Racing/shepherd_bms #132: Make sending the charger CAN message use a union rather than the funky bitshfiting

nwdepatie, 2023-02-06:
**What are you proposing?**

**What files/components does this concern?**

**What would this improve over how it is now?**

**Estimated time commitment**

dyldonahue, 2023-02-09:
(compute.cpp line 58)

@mccauleyma wth is going on with this. There's already a union set up presumably with the correct size for each element, what came up that caused the need for all the bit shifting in the CAN msg itself.

dyldonahue, 2023-02-09:
@nwdepatie matt said that this was put in place because when testing the union seemed to break for a while. 

Given that, I think we should cautiously try removing all the funky bit shifting at a time where we can test it right away to see if our union works as expected or not. On hold until then

closed 2023-06-10 by dyldonahue